openmv/openmv-ide #1: Actual IDE

edge0701, 2016-07-23:
Are you planning to release your current version of IDE open source? At the moment it just builds QT Creator, unless I am missing something? Just received my board and only find binaries...

edge0701, 2016-07-23:
Ah. I found the current IDE in OpenMV/openmv/usr! Sorry about that!

closed 2016-07-23 by edge0701

kwagyeman, 2016-07-23:
Um, that's the old IDE. The new IDE is here: https://github.com/openmv/openmv-ide

The new IDE is built on top of QtCreator. You have to use the openmv branch of the QtCreator repo.

https://github.com/openmv/qt-creator/tree/f6fe0d2f8c9a9b801a4ffa99927205ecdf20bf7b

Lots of stuff was modified in the code. All new changes are here: https://github.com/openmv/qt-creator/tree/f6fe0d2f8c9a9b801a4ffa99927205ecdf20bf7b/src/plugins/openmv

edge0701, 2016-07-24:
I was on the wrong branch as I had pulled manually. All working.

Thanks for your help. And great work.