openmv/openmv-ide #2: 1.0 often crashes on OSX when clicking "disconnect"

petri, 2016-08-23:


kwagyeman, 2016-08-23:
Thank's for using the software! I have a new release almost ready where I completely redid the serial code. Hopefully that will come out this weekend. When you test out the new release let me know if the problem still exists.

Yeah, again, I redid the serial code to make sure stuff like this doesn't happen. However, I've noticed that when the camera has a problem and it's USB driver crashes this causes whatever program opened the serial port to lock up to. Anyway, disconnecting the camera usually fixes this.

Until I get get Ibrahim to fix the serial library stuff like this will remain an issue. The OpenMV Cam serial code is... poor in my opinion... just really fragile.

petri, 2016-09-05:
Still an issue on 1.1.0.

kwagyeman, 2016-09-05:
Does this happen 100% of the time?

petri, 2016-09-07:
I think it's worth adding a note to the OSX download that you only support OSX El Capitan. I did not remember I was still on OSX Yosemite. Upgraded to El Capitan overnight and seems like things are a lot more stable, including this issue.

petri, 2016-09-08:
Seems the disconnect crash now only happens after the slow REPL (which is also followed by the start/stop button state fluctuations).

kwagyeman, 2016-11-10:
Does the IDE work now? See v1.3.

petri, 2016-11-15:
Seems to work ok now, both when there is no script running and when there is.

closed 2016-11-15 by petri