openmv/openmv-ide #3: 1.0.0 terminal output is sometimes veeeery slow on OSX

petri, 2016-08-23:
For no apparent reason, sometimes output printed from the program is echoed to the IDE very, very slowly, a character at a time (a second or so between each character).

As far as I can tell, this is not because the program itself causes such slowness.

kwagyeman, 2016-08-23:
Thank's for using the software! I have a new release almost ready where I completely redid the serial code. Hopefully that will come out this weekend. When you test out the new release let me know if the problem still exists.

(I have no clue why this would happen. Whatever the case, not much I can do if it were happening.)

petri, 2016-09-05:
Still an issue on 1.1.0.

petri, 2016-09-07:
I think it's worth adding a note for Mac users on the IDE download page that you only support OSX El Capitan. I did not remember I was still on OSX Yosemite. Upgraded to El Capitan overnight and seems like things are a lot more stable, including this issue.

I will do some testing to find out if there still are issues.

kwagyeman, 2016-09-07:
I got an update coming soon that will switch to using 64-byte packets which should fix a lot of issues.

kwagyeman, 2016-11-10:
Does the IDE work now? See v1.3.

petri, 2016-11-15:
Quick testing on 1.3 OSX El Capitan indicates this is no longer a problem.

closed 2016-11-15 by petri