openmv/openmv-ide #4: After stopping, the "stop" button sometimes returns to enabled state

petri, 2016-08-23:
This happens on 1.0.0. on OSX quite often. The button is disabled first, but after a second or two, it returns to being enabled. So both "start" and "stop" buttons end up being enabled.

It also seems almost like the stop events are buffered somewhere and get processed by camera after the next time a program is run, prematurely halting it.

kwagyeman, 2016-08-23:
Thank's for using the software! I have a new release almost ready where I completely redid the serial code. Hopefully that will come out this weekend. When you test out the new release let me know if the problem still exists.

petri, 2016-09-05:
Update on 1.1.0: There seems to be now just one button. What happens is that sometimes after pressing start, it more or less rapidly starts switching back and forth between red and green on its own.

kwagyeman, 2016-09-05:
Okay, so, this happens when I get timeouts for commands. In general, it's taking the OpenMV Cam forever to respond to serial traffic. I poll the script started state to determine if a script is running or not.

Increasing the serial timeout somewhat fixes this issue. But, I don't really have direct control over what's causing this problem. I'll release a new IDE very soon with a bunch of fixes for this.

I don't.... really know what to do to make this better however. It's ridiculous that the Cam takes over 500 ms to respond.

petri, 2016-09-05:
I see. Perhaps all we can do is "damage control" then, by accurately informing the user of what is happening. If we can remove unpredictable behavior (from user perspective), that goes a long way. Would it be ok to add a spinner to the IDE UI to indicate we're waiting for the cam? Or something.

Also, I noticed that quite often, it seems that the serial REPL is showing stuff that is old. It feels like there's a serial FIFO buffer that's been filled up during previous script execution, or another operation, and then at some later point starts to spew out its contents.

kwagyeman, 2016-09-05:
The issue is with split packets I believe. The OpenMV Cam firmware was written for libusb and was ported over for a normal serial port. Anyway, the firmware assumes serial commands come as one packet. And if they don't it breaks. I'm trying to work around this, but, on different OSes things work differently. Its really all highly variable. Until the firmware is fixed this problem won't go away.

Anyway, in the mean time, I'll make the IDE re-try commands that fail once more. This should help.

petri, 2016-09-07:
I think it's worth adding a note for Mac users on the IDE download page that you only support OSX El Capitan. I did not remember I was still on OSX Yosemite. Upgraded to El Capitan overnight and seems like things are a lot more stable, including this issue.

I will do some testing to find out if there still are issues.

kwagyeman, 2016-11-10:
Does the IDE work now? See v1.3.

petri, 2016-11-15:
Seems no longer an issue with IDE 1.3 on OSX El Capitan.

closed 2016-11-15 by petri