openmv/openmv-ide #5: Camera crashes if IDE overwrites the running script (1.x.x on OSX El Capitan)

petri, 2016-09-08:
Steps to reproduce:

1. connect camera
2. start a script
3. while the script is still running, overwrite the (running) script (via IDE)

There will usually be different errors, apparently depending on the actual changes to the script. The only way to recover is to physically disconnect-reconnect the USB.

The crash does not always happen - it seems the more there are changes to the script, the higher the likelihood of crash.

I would suggest disabling the save button while the script runs.

kwagyeman, 2016-09-09:
Got a new version coming soon. But, um, which save script button? The save script to the computer or camera?

I'm pretty sure you're having all these issues because of the serial driver crashing because of the way the OpenMV Camera firmware works. I'm hopping the the upcoming switch to 64-byte packets in the IDE will fix this.

petri, 2016-09-12:
Ok. I meant a script stored on (and running from) the SD card attached to the camera.

kwagyeman, 2016-09-13:
Sounds like a firmware issue.

Can you post this issue in the openmv repo? There's nothing sepcial going on in the IDE to update the script while running. It's just saving a file to the SD card using the USB flash drive access.

kwagyeman, 2016-09-13:
Oh, so, I got busy with business stuff this weekend. Wasn't able to finish the IDE updates. Hopefully I can get back to that tomorrow.

kwagyeman, 2016-11-10:
Does the IDE work now? See v1.3.

petri, 2016-11-15:
With 1.3, the following now happens on OSX El Capitan:

- when using the special Tools -> "save open script to OpenMV cam" menu item from the IDE, nothing happens; it seems the IDE refuses to save the script

- when instead attempting to save the script via the File -> Save menu item in IDE, the IDE pops up an error message (OSError; Assertion failed or OSError; unsupported format)

kwagyeman, 2016-11-15:
Hi, can you close this issue and open up for the OpenMV repo as a firmware issue? I can't do anything other than disable the ability to save a script while one is running in the IDE. However, it sounds like the cam should be able to handle this situation.

closed 2016-12-28 by kwagyeman