openmv/openmv-ide #8: Failed to load core: Plugin initialization failed: No themes found in installation.

edge0701, 2017-01-30:
Hi OpenMV,

I read about your new 1.4.0 release today with QRCode implementation which I wanted to try out!
Checked out from git, init and update submodules, qmake and make all run fine. (Not using make.py as I don't want to produce the installer). But I get the error as described in issue title, full message is:
```
Warning: No themes found in installation:  "/home/[redacted]/src/openmv/openmv-ide-master/share/qtcreator/themes"
Failed to load core: Plugin initialization failed: No themes found in installation.
```
Indeed, there are no themes in the share directory, only translations.
(I have tried both master and v1.4.0 tagged release from git).

kwagyeman, 2017-01-30:
Um, the installer is already built for linux. Maybe use that?

But, probably you’re on the wrong branch. Make sure to checkout the openmv branch of the qtcreator submodule. The master branch for openmv-media and openmv-ide is fine.

Then, just built using the qtcreator.pro file in the qtcreator submodule. make.py is for the installer.

edge0701, 2017-01-31:
I don't see any instructions for this either documented or in make.py. It would be good to make this clear.
I started from scratch and cloned openmv-ide, ran ```git submodule update --init --recursive``` then went into qt-creator and ran ```git checkout openmv```.

As an aside, I have had issues creating the installer on ArchLinux due to QtIFW location.

Thanks for your help. Much appreciated.

closed 2017-01-31 by edge0701

kwagyeman, 2017-01-31:
Um, hasn't really been a priority to document this all. Probably should. Note that even though you managed to build the installer on linux you really need to use a super old version of linux when doing so if you want to distribute the application.

Ton-o-steps to set the build system up for that (should have written them down).