openmv/openmv-ide #9: qtcreator.pro missing

deshipu, 2017-02-14:
I'm trying to build this, I figured out where to put all the deps, but now when I run `make.py` I get:

```
$ python make.py 
Cannot find file: ../qt-creator/qtcreator.pro.
Make Failed...
```

It seems that file is missing from the repository.

kwagyeman, 2017-02-14:
You need to install the qtcreator sub repo. You have to init the submodule.

deshipu, 2017-02-14:
Ah, I see, thanks!

closed 2017-02-14 by deshipu

kwagyeman, 2017-02-14:
Also make sure to checkout the openmv branch of the submodule. Not sure why git doesn't see this. It's in the submodule spec file.

kwagyeman, 2017-02-14:
Um, one more thing. Don't use the make.py script. That's for release. Just load the qtcreator.pro file directly in QtCreator and it will create the build dir, etc. for you and you can start hacking.

deshipu, 2017-02-14:
Excellent, thanks!

deshipu, 2017-02-14:
Hmm, I'm getting ":-1: error: Unknown module(s) in QT: qml" both in qtcreator and with make.py, and I'm not sure what I need to install -- there are 100s of packages with "qml" in their name...

kwagyeman, 2017-02-14:
Um, qml shouldn't be showing up.  Pretty sure I unlinked that. You might have installed the qml submodule in qtcretor. You don't want that. Just destroy the repo for that in the qctcreator repo.

kwagyeman, 2017-02-14:
Once the repo for that is destroyed then the build script shouldn't pick it up. (the qtcreator.pro file that is).

deshipu, 2017-02-14:
Not sure what you mean here. How do I destroy that repo?

kwagyeman, 2017-02-14:
Just delete the dir that gets inited from the qtcreator .submodule file.

kwagyeman, 2017-02-14:
https://github.com/openmv/qt-creator/blob/1793b0b5cdec2d929e3140ba628cf8124fa12143/.gitmodules

kwagyeman, 2017-02-14:
src/shared/qbs

deshipu, 2017-02-14:
Now I get: :-1: error: Unknown module(s) in QT: help, and deleting src/shared/help didn't help.

kwagyeman, 2017-02-14:
You only wanted to delete qbs. Um, how did you install Qt?

deshipu, 2017-02-14:
apt-get install qtcreator

kwagyeman, 2017-02-14:
Oh, yeah, don't use that.  Install the SDk from qt online. Or, also do "apt-get install qt".

QtCreator contains all the tools to build Qt code but not any of the libraries.

I'd just install everything from this: https://www.qt.io/download-open-source/

deshipu, 2017-02-14:
E: Unable to locate package qt

I'd rather not install non-packaged stuff on my system. I guess I will need a virtual machine for this.

Thank you for your help.

kwagyeman, 2017-02-14:
Oh, no, it's safe to do. The qt installer just unpacks everything to your home dir. It doesn't modify any paths or anything. If you want to remove it you just have to delete the home dir.

Anyway, what are you trying to do?

kwagyeman, 2017-02-14:
https://wiki.qt.io/Install_Qt_5_on_Ubuntu

deshipu, 2017-02-14:
I'm presenting a talk about MicroPython on a conference on Friday, and I wanted to also show OpenMV there -- for that, I wanted to see how the newest version of the IDE works.

kwagyeman, 2017-02-14:
Okay, uh, but, why do you need to build it versus just showing off the code?

deshipu, 2017-02-14:
I just wanted to run the most recent version.

kwagyeman, 2017-02-14:
Oh, well, the only changes I made were to add AprilTag generating code from what has been released. You don't have an M7 cam so you can't use that just yet. Anyway, I can make a quick build for you if you like. Otherwise, you'll need to install Qt on your system.

kwagyeman, 2017-02-14:
(And yes, I understand how much building the IDE is a pain. Trust me that for actually building the release versions I had to go through even more pain since you have to build them on Ubuntu 12).

deshipu, 2017-02-14:
I actually wanted to do it now, so that I can also follow the development as it happens and see what you are working on, without having to wait for releases. Of course, this is not important, and I can as well just show the released version at the talk (in fact, I probably should, as it will be less likely to crash). I guess it's just the open source habits.

kwagyeman, 2017-02-14:
Okay, well, follow the link I posted above to build on linux then. For just developing you don't have to do anything more than that.

deshipu, 2017-02-14:
Thanks again!

kwagyeman, 2017-02-14:
Oh, BTW. I finally have time to fix the syntax highlighting and autocomplete. I'd working on that next. The next release of the IDE should be the first one with no obvious unfinished parts.

deshipu, 2017-02-14:
Great to hear. It's amazing how much work you are putting into this. Makes me feel guilty for not finishing that hexapod robot with openmv on it. But it's not abandoned, just shelved...