openmv/openmv-ide #11: Cannot import a custom file in the main folder

AlexisTM, 2017-06-02:
Dear, 

When I am trying to import the Tag.py file (import Tag) which I putted inside the storage of the camera, the IDE complains it cannot find the file but is looking in the wrong path. 

> Could not open "/media/alexis/0D65-A963Tag.py" for reading. Either the file does not exist or you do not have the permissions to open it.

The correct path is `/media/alexis/0D65-A963Tag.py`. 

Cannot import a file in the folder. Error from OpenMVIDE

AlexisTM, 2017-06-02:
Note that this Error comes from the IDE while the program is actually run on the camera itself. (As I get the error "module as not attribute hexversion" error due to the Python check from the file I was writing.)

kwagyeman, 2017-06-02:
Hi, OpenMV IDE does not process imported files in code. That's done on the MicroPyhton board itself. So... If you are having an import error from a script it's not the IDEs fault.

Can you give me some more context on what you are trying to do? I'm not following what's going on.

I can get a fix into the next release this weekend if you let me know in a lot of detail what the problem is.

kwagyeman, 2017-06-02:
If a script is in a folder then the import rules change. You can import a script by name when it's in the same dir as the current script. Otherwise you have to create a folder with the module name and then put an init.py<http://init.py> script in that folder. Google MicroPython import rules for a good article about all of this.

AlexisTM, 2017-06-02:
Indeed, the program is running fine on the camera itself, but still have an error with a local path.

To reproduce it, you can import a file called "tag.py" with any content like: 

    print("Up and running")

In OpenMV IDE, create a file anywhere with the content:

    import tag

Connect to the camera, click start and will see an error pointing a local path with a missing /.

I use:
- OpenMV 7
- Last version of OpenMV IDE
- Ubuntu 14.04

kwagyeman, 2017-06-07:
After copying the file over onto the camera you need to reset the camera. MP can't see files written by the host PC immediately. Tools -> Reset OpenMV Cam.

closed 2017-06-07 by kwagyeman