openmv/openmv-ide #15: wifi programming

goyzhang, 2018-04-09:


kwagyeman, 2018-04-09:
Oh, wow, this feature is not in use yet... Um, we don't have wifi programming support yet so please ignore this.

The port is ABD1.

goyzhang, 2018-04-09:
I see, thank you.  BTW what did you mean by ABD1, shouldn't that be a ipv4 port number?

kwagyeman, 2018-04-09:
#define OPENMVCAM_BROADCAST_PORT 0xABD1

It's a UDP broadcast port we bind to that the OpenMV Cam will send broadcast UDP packets on. WiFi programming has been completed yet. However, we've basically got the infrastructure for it. Random other things have been delaying it's implementation...

That said, we're going to release soon, maybe I'll do this right afterwards. I've been itching to add some big features...

closed 2018-04-09 by kwagyeman