openmv/openmv-ide #17: Is current openmv-ide compatible with Python 3.6.5 ?

jiapei100, 2018-05-09:
Is current openmv-ide compatible with Python 3.6.5 ?

> 
> make[2]: Entering directory '/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations'
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_cs.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_cs.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_de.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_de.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_fr.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_fr.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_ja.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_ja.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_pl.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_pl.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_ru.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_ru.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_sl.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_sl.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_uk.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_uk.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_zh_CN.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_zh_CN.qm
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations/qtcreator_zh_TW.qm /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/share/qtcreator/translations/qtcreator_zh_TW.qm
> make[2]: Leaving directory '/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share/qtcreator/translations'
> make[1]: Leaving directory '/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/share'
> cd bin/ && ( test -e Makefile || /opt/Qt/Current/gcc_64/bin/qmake -o Makefile /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/bin/bin.pro -spec linux-g++ ) && make -f Makefile install
> make[1]: Entering directory '/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/bin'
> /opt/Qt/Current/gcc_64/bin/qmake -install qinstall -exe openmvide.sh /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install/bin/openmvide.sh
> make[1]: Leaving directory '/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/bin'
> python -u /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/scripts/deployqt.py -i "/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/install" "/opt/Qt/Current/gcc_64/bin/qmake"
>   File "/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/scripts/deployqt.py", line 44
>     print "Usage: %s <creator_install_dir> [qmake_path]" % os.path.basename(sys.argv[0])
>                                                        ^
> SyntaxError: invalid syntax
> Makefile:567: recipe for target 'deployqt' failed
> make: *** [deployqt] Error 1
> Make Failed...
>

jiapei100, 2018-05-10:
python 2.7 builds successfully, but failed to install?

> Files read from disk: 1099
> Archive size: 65048385 bytes (63 MiB)
> Everything is Ok
> python -u /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/scripts/packageIfw.py -i "" -v 1.3.0 -a "/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/build/openmv-ide-linux-x86_64-1.3.0-installer-archive.7z" openmv-ide-linux-x86_64-1.3.0 && python -u /media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/scripts/sign.py "openmv-ide-linux-x86_64-1.3.0.run"
> Traceback (most recent call last):
>   File "/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/scripts/packageIfw.py", line 155, in <module>
>     main()
>   File "/media/jiapei/Data/Downloads/machinevision/openmv/openmv-ide/qt-creator/scripts/packageIfw.py", line 90, in main
>     raise Exception('Installer framework location not specified (--installer-path)!')
> Exception: Installer framework location not specified (--installer-path)!
> Makefile:578: recipe for target 'installer' failed
> make: *** [installer] Error 1
> Make Failed...
>

kwagyeman, 2018-05-10:
You need to add install the qt installer framework on your system. This may have been the reason for the previous none type error.

closed 2018-05-10 by kwagyeman

jiapei100, 2018-05-10:
@kwagyeman 
Oh, yes... I actually installed my own QT already... That might be the reason...
Anyway, I'm running **openmv** !! But, the camera is **NOT** able to be detected at all. Ubuntu 18.04 . I guess it has something to do with the hardware. My **cmucam5** is NOT working either.

kwagyeman, 2018-05-10:
Hi, you have to install qt installer framework. Also, we don't support the CMUcam5.

Please use the binaries posted online if you'd like to use your OpenMV Cam with OpenMV IDE.