openmv/openmv-ide #18: Menus in OpenMV IDE are black on black text

able-archer, 2018-05-16:
I just installed the OpenMV IDE on a fresh Kubuntu 18.04.
I'm using the Breezy Dark theme, which is one of the 3 default themes in Kubuntu.
The IDE works fine, but the text in the menus are black on a very dark background, making it nearly impossible to read.
The menu's are fine if I choose a light theme in Kubuntu, but the dark one is easier on the eyes.

Could you create a setting to change the color of the font in the menus, or detect a darktheme usage in some way?

[https://i.gyazo.com/ca1154a363f662f7c66349205f688b05.png](url)
[https://i.gyazo.com/27841e7b9a288cb406970550028c79e1.png](url)

kwagyeman, 2018-05-16:
Thanks for creating a bug tracker on this. I'll try to figure out why the IDE is getting this wrong.

kwagyeman, 2018-06-29:
I've fixed this for the next release.

closed 2018-06-29 by kwagyeman