openmv/openmv-ide #19: Fail to open .py file in openmv-ide?

jiapei100, 2018-06-09:
```
$ MimeDatabase: Error loading :/qt-project.org/qmime/freedesktop.org.xml
Cannot open :/qt-project.org/qmime/freedesktop.org.xml: No such file or directory
QFSFileEngine::open: No file name specified
Gtk-Message: 02:12:34.925: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 02:13:08.435: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 02:13:16.724: GtkDialog mapped without a transient parent. This is discouraged.
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&) unable to determine mime type of .../firsttest.py/. Falling back to text/plain
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&): unable to find an editor factory for the file '.../firsttest.py', editor Id ''.
^C
[1]+  Done                    openmvide
```

kwagyeman, 2018-06-09:
Can you provide the operating system? And the file? We can't exactly reproduce with that trace.

jiapei100, 2018-06-10:
@kwagyeman 

1. OS: Ubuntu 18.04

2. OpenMV 1.9.0

About OpenMV IDE 1.9.0
By: Ibrahim Abdelkader & Kwabena W. Agyeman
GNU GENERAL PUBLIC LICENSE
Copyright (C) 2017 OpenMV, LLC
This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more details.
Questions or Comments?
Contact us at openmv@openmv.io.
Credits
OpenMV IDE English translation by Kwabena W. Agyeman.

3. Qt: 5.11.0

kwagyeman, 2018-06-10:
Super, now the file?

jiapei100, 2018-06-18:
Check this please ...
[http://forums.openmv.io/viewtopic.php?f=3&t=821](http://forums.openmv.io/viewtopic.php?f=3&t=821)

kwagyeman, 2018-07-01:
Closing this.

closed 2018-07-01 by kwagyeman