openmv/openmv-ide #24: Autosetting resolution for video converter

Hydrorion, 2018-07-13:
The OpenMV CAM M7 seems not to be powerfull enought to handle the reading of imported videos with better quality of qvga. Since the videos we import with the new converter will propably be in better quality, and that the converter take a bit a while to convert the videos, is this possible to set the quality to QVGA so we don't have to convert on internet ourselve ? It will be really great !

Just one thing about the converter : it seems like we can't  stop the converter when it is processing. Each time we click on "aboard" the dialog box reappear and block all the OpenMV IDE. The first time i use it i converted a big file and it taken to much time so i decided to stop but i had to stop my computer :/

Otherwise it's a really good new feature, thank for it !

kwagyeman, 2018-07-13:
Will do, note that the converter runs in another terminal window. Just click the X button on that window to stop it. Also, remember the OpenMV Cam is a microcontroller. We only have 512KB of RAM on the M7.

kwagyeman, 2019-04-09:
Hi, I've added this feature now.

closed 2019-04-09 by kwagyeman