openmv/openmv-ide #25: qt-creator pull request?

jiapei100, 2018-07-14:
It looks openmv-ide and qtcreator are of two different git repos?
Anyway, I cannot pull request directly from openmv-ide directory, for some modifications must be made under subfolder: qt-creator.

```
- /src/plugins/coreplugin/dialogs/readonlyfilesdialog.cpp
+#include <QButtonGroup>

- /src/plugins/coreplugin/find/findtoolbar.cpp
+#include <QAbstractItemView>
```

- scripts/deployqt.py
For **ALL print**, add parenthesis ()

etc...

kwagyeman, 2018-07-14:
I don't understand this error.

jiapei100, 2018-07-16:
@kwagyeman 

I mean, when I tried to compile **openmv-ide** from source with **python3**, I got some **ERROR** messages. So, the above 3 points are partial solutions...

kwagyeman, 2019-04-09:
Hi, you would have to create a fork of both repos and then submit a PR for each.

closed 2019-04-09 by kwagyeman