openmv/openmv-ide #27: Two ERRORs if I build openmv-ide from source...

jiapei100, 2018-07-15:
Environment: **Ubuntu 18.04 + gcc 7.3.0 + Python 3.6.5**

After having built openmv-ide from source from scratch, I had almost all functionalities enabled for my **OpenMV3 R2** . But, still with the following **2 ERROR** messages:

1. File **packagelfw.py** around line **144**

```
ifw_call = [os.path.join(ifw_location, 'bin', 'binarycreator'), '-c', os.path.join(out_config_dir, config_name), '-p', out_packages_dir, installer_name, '--offline-only' ]
```
gives the following **ERROR** message:

```
Files read from disk: 1381
Archive size: 103526650 bytes (99 MiB)
Everything is Ok
python -u ....../openmv/openmv-ide/qt-creator/scripts/packageIfw.py -i "None" -v 2.0.0 -a "....../openmv/openmv-ide/build/openmv-ide-linux-x86_64-2.0.0-installer-archive.7z" openmv-ide-linux-x86_64-2.0.0 && python -u ....../openmv/openmv-ide/qt-creator/scripts/sign.py "openmv-ide-linux-x86_64-2.0.0.run"
Cleaning up...
Done.
Traceback (most recent call last):
  File "....../openmv/openmv-ide/qt-creator/scripts/packageIfw.py", line 155, in <module>
    main()
  File "....../openmv/openmv-ide/qt-creator/scripts/packageIfw.py", line 147, in main
    subprocess.check_call(ifw_call, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.6/subprocess.py", line 286, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 267, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'None/bin/binarycreator': 'None/bin/binarycreator'
Makefile:590: recipe for target 'installer' failed
make: *** [installer] Error 1
Make Failed...
```

2. Fail to open **.py** files:
```
$ MimeDatabase: Error loading :/qt-project.org/qmime/freedesktop.org.xml
Cannot open :/qt-project.org/qmime/freedesktop.org.xml: No such file or directory
QFSFileEngine::open: No file name specified
Gtk-Message: 10:55:03.237: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 10:55:12.926: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 10:55:34.445: GtkDialog mapped without a transient parent. This is discouraged.
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&) unable to determine mime type of ....../OpenMV/PeopleCounterContours.py/. Falling back to text/plain
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&): unable to find an editor factory for the file '....../OpenMV/PeopleCounterContours.py', editor Id ''.
Gtk-Message: 10:56:22.894: GtkDialog mapped without a transient parent. This is discouraged.
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&) unable to determine mime type of ....../OpenMV/PeopleCounterContours.py/. Falling back to text/plain
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&): unable to find an editor factory for the file '....../OpenMV/PeopleCounterContours.py', editor Id ''.
Gtk-Message: 10:56:30.576: GtkDialog mapped without a transient parent. This is discouraged.
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&) unable to determine mime type of ....../OpenMV/PeopleCounterContours_nogui.py/. Falling back to text/plain
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&): unable to find an editor factory for the file '....../OpenMV/PeopleCounterContours_nogui.py', editor Id ''.
Gtk-Message: 10:56:38.307: GtkDialog mapped without a transient parent. This is discouraged.
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&) unable to determine mime type of ....../OpenMV/head_counter.py/. Falling back to text/plain
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&): unable to find an editor factory for the file '....../OpenMV/head_counter.py', editor Id ''.
Gtk-Message: 10:56:47.601: GtkDialog mapped without a transient parent. This is discouraged.
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&) unable to determine mime type of ....../OpenMV/PeopleCounterContours_nogui.py/. Falling back to text/plain
static Core::EditorManager::EditorFactoryList Core::Internal::EditorManagerPrivate::findFactories(Core::Id, const QString&): unable to find an editor factory for the file '....../OpenMV/PeopleCounterContours_nogui.py', editor Id ''.
^C
[1]+  Done                    ./openmvide.sh
```

kwagyeman, 2018-07-15:
It says the error in the error message... For the first one it can't find an executable normally available in the Qt installer frame work for her which you have to install your self. The installer frame work is a separate download and install.

As for the second error I have no idea still. Is previously mentioned... I build using Qt 5.7. you may wish to downgrade to that.

closed 2018-07-15 by kwagyeman

jiapei100, 2018-07-16:
Yup, you are right. I'm building **openmv-ide** against **Qt 5.11.1** ....

jiapei100, 2018-07-16:
The 2nd **ERROR** is now solved by **modifying** **line 805** of file 

> qt-creator/scr/libs/utils/mimetypes/mimeprovider.cpp

from 

```
allFiles.append(QLatin1String(":/qt-project.org/qmime/freedesktop.org.xml"));
```

to

```
allFiles.append(QLatin1String("/usr/share/mime/packages/freedesktop.org.xml"));
```

jiapei100, 2018-07-16:
@kwagyeman 

an executable? What executable?

```
'None/bin/binarycreator'
```

Is it looking for the binary of **qtcreator** ? Or, **qmake**? or any other executable file?

> It says the error in the error message... For the first one it can't find an executable normally available in the Qt installer frame work for her which you have to install your self. The installer frame work is a separate download and install.