openmv/openmv-ide #29: Why Clang is adopted? I'm using Ubuntu 18.04.1...

jiapei100, 2018-09-05:
Ubuntu 18.04.1
Python 3.6.5
LLVM_INSTALL_DIR=/usr/lib/llvm-6.0

Why Clang is adopted? It seems **ONLY** **darwin** will invoke Clang. I prefer using GCC. And I'm NOT using rpi, but a laptop....

```
copying libclang...
/usr/lib/llvm-6.0/lib/libclang.so -> ....../openmv-ide/build/install/lib/qtcreator
/usr/lib/llvm-6.0/lib/libclang.so.1 -> ....../openmv-ide/build/install/lib/qtcreator
/usr/lib/llvm-6.0/bin/clang -> ....../openmv-ide/build/install/lib/qtcreator
removing libclang RPATHs...
open: Permission denied
elf_open: Permission denied
Traceback (most recent call last):
  File "....../openmv-ide/qt-creator/scripts/deployqt.py", line 340, in <module>
    main()
  File "....../openmv-ide/qt-creator/scripts/deployqt.py", line 325, in main
    deploy_libclang(install_dir, os.environ["LLVM_INSTALL_DIR"], chrpath_bin)
  File "....../openmv-ide/qt-creator/scripts/deployqt.py", line 249, in deploy_libclang
    subprocess.check_call([chrpath_bin, '-d', targetfilepath])
  File "/usr/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/chrpath', '-d', '....../openmv-ide/build/install/lib/qtcreator/libclang.so']' returned non-zero exit status 1.
Makefile:581: recipe for target 'deployqt' failed
make: *** [deployqt] Error 1
Make Failed...
```

Any suggestions?

Cheers
Pei

kwagyeman, 2018-09-05:
I use gcc to compile OpenMV IDE. I don't know what problem you are having.

jiapei100, 2018-09-05:
@kwagyeman 

The ONLY thing I've done is to 
```
python make.py
```

And, it automatically look for **CLANG**, which is out of my expectation....

jiapei100, 2018-09-05:
@kwagyeman 

Alright, I made some progress. From the 1st post, it is an **ERROR** of **Permission denied** .
I'm **NOW** using 
```
sudo python make.py
```
which seems to progress a bit. However, from the following **ERROR** message, it seems I need to specify the installation directory???

```
7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=en_CA.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz (506E3),ASM,AES-NI)

Scanning the drive:
108 folders, 1575 files, 626472928 bytes (598 MiB)                      

Creating archive: ....../openmv-ide/build/openmv-ide-linux-x86_64-2.1.0-installer-archive.7z

Items to compress: 1683

                                                                            
Files read from disk: 1575
Archive size: 106855416 bytes (102 MiB)
Everything is Ok
python -u ....../openmv-ide/qt-creator/scripts/packageIfw.py -i "None" -v 2.1.0 -a "....../openmv-ide/build/openmv-ide-linux-x86_64-2.1.0-installer-archive.7z" openmv-ide-linux-x86_64-2.1.0 && python -u ....../openmv-ide/qt-creator/scripts/sign.py "openmv-ide-linux-x86_64-2.1.0.run"
Cleaning up...
Done.
Traceback (most recent call last):
  File "....../openmv-ide/qt-creator/scripts/packageIfw.py", line 155, in <module>
    main()
  File "....../openmv-ide/qt-creator/scripts/packageIfw.py", line 147, in main
    subprocess.check_call(ifw_call, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.6/subprocess.py", line 286, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 267, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'None/bin/binarycreator': 'None/bin/binarycreator'
Makefile:590: recipe for target 'installer' failed
make: *** [installer] Error 1
Make Failed...
```

Cheers
Pei

jiapei100, 2018-09-05:
It seems it's **ONLY** the last step **to install the built package** is buggy.
So, I just re-run the last command, specify the installer-path by **-i "/opt/openmv"**, and remove **binarycreator** in file **packageIfw.py**...

Still, failed to install as follows... 

```
$ sudo python -u ....../openmv-ide/qt-creator/scripts/packageIfw.py -i "/opt/openmv" -v 2.1.0 -a "....../openmv-ide/build/openmv-ide-linux-x86_64-2.1.0-installer-archive.7z" openmv-ide-linux-x86_64-2.1.0 && python -u ....../openmv-ide/qt-creator/scripts/sign.py "openmv-ide-linux-x86_64-2.1.0.run"
Cleaning up...
Done.
Traceback (most recent call last):
  File "....../openmv-ide/qt-creator/scripts/packageIfw.py", line 155, in <module>
    main()
  File "....../openmv-ide/qt-creator/scripts/packageIfw.py", line 147, in main
    subprocess.check_call(ifw_call, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.6/subprocess.py", line 286, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 267, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: '/opt/openmv/bin/'
```

I'm NOW going to **copy and paste** everything manually, since it's already been successfully built entirely.

Cheers
Pei

kwagyeman, 2018-09-06:
It can't find binary creator from QtIFW

closed 2019-04-09 by kwagyeman