openmv/openmv-ide #33: translate MicroPython runtime error, such as SyntaxError, Memory Error

kidswong999, 2019-02-02:
Is there any way to translate MicroPython runtime error? such as SyntaxError, Memory Error？

[image]

kidswong999, 2019-02-02:
If direct replacement into translation can be ambiguous, then we can add a line of translation below.

kidswong999, 2019-02-02:
I can organize and translate them.😝

kwagyeman, 2019-02-02:
Yeah, I can do it in the IDE. I just need a lookup table for the translations. Qt provides this lut function for free. So, I just need a translation in Chinese for all the strings in the firmware.

kwagyeman, 2019-02-02:
Um, I think you can use a command line tool to dump all the strings in the .elf file.

kwagyeman, 2019-02-02:
Um, once you find a string you just need to add it here:

https://github.com/openmv/qt-creator/blob/4c593eec91a957c564e1da426611120d53c3bce2/share/qtcreator/translations/qtcreator_zh_TW.po

Lines with # are comments. msgctxt is just a comment basically too. Only the msgid and msgstr matter. However, I'd put all 3 fields, msgctxt, msgid, and msgstr in a row for new entries.

Translation is just a big lookup based on the input text being matched to an id and replaced with the new str. If there's no match then there's no translation.

kwagyeman, 2019-02-02:
I have to only change a line of code in the IDE to then pull from these translations once they are added. If you do the work of finding and translating all the strings I'll do the rest for other languages.

kidswong999, 2019-02-13:
https://github.com/openmv/qt-creator/pull/7
The error message should have both English and local languages, because there are many English content when searching.

kwagyeman, 2019-02-13:
Just a note, I'm not sure exactly how I will handle the replacement chars. They will not be in the string sent to me which prevents direct translation. I'll have to somehow detect during translation what the replacement char would be and then translate it. Basically, I need to do a wild card search on the translation DB. I don't know if this is possible.

kwagyeman, 2019-04-11:
Hi, I've added translation support to incoming error messages. However, I can't deal with the %1 stuff. So, all possible error variants of strings will need to be added for translation to work. The translation method is just a hash table lookup. So, it needs to find a direct match to the error message. The %1 things in the error messages can be replaced with anything breaking this.

closed 2019-04-11 by kwagyeman