openmv/openmv-ide #35: Failed to install openmvide under Raspberry Pi 3B (ubuntu mate 16.04.2)

BreederBai, 2019-02-21:
I am getting an error when I execute "./setup.sh"
[image]
Display libgles1-mesa-dev dependent libgles1-mesa version is wrong, need 12.0.6-0bantu.16.04.1 but 18.0.5-0ubuntu0~16.01.1 is about to be installed

kwagyeman, 2019-04-16:
Hi, I installed the IDE on a fresh Raspberry Pi 3 using the latest image (as of April 15th) and it works. I will not be debugging all the different types of linux errors that can happen.

Part of the FUN of linux is doing this yourself.

In the future. I will accept PRs for fixes to the install scripts for these issues. But, I cannot debug every possible issue.

closed 2019-04-16 by kwagyeman

BreederBai, 2019-04-18:
@kwagyeman Hello, I am using the operating system when ubuntu mate 16.04, what is the system you are using?

kwagyeman, 2019-04-18:
Hi, I've only tried using the default install of Raspbian on the Pi.

Regarding the issue above... In my install script here I don't specify the version of the software to be installed. This means your package system is setup with problems. E.g. here's what commands the install script runs: https://github.com/openmv/openmv-ide/blob/e0e2c454c5b967851274ddeed316cf4d8f9dfa51/make.py#L164

BreederBai, 2019-04-18:
Hello, which version of the system are you using, can you give me a link to download it.

kwagyeman, 2019-04-18:
This one: https://www.raspberrypi.org/downloads/raspbian/

First download link.

BreederBai, 2019-04-22:
@kwagyeman Thank you