openmv/openmv-ide #37: Serial Terminal - Save

t1jump, 2019-03-15:
Can a Save option for the Serial Terminal/Log file be added (i.e. Save button next to existing Clear button)?

kwagyeman, 2019-03-15:
Can you go over the type of flow you'd like for this? How do you see it being used? Etc. It's helpful if I cna know exactly what you want.

t1jump, 2019-03-15:
Our application is use in RoboCup Rescue Robot League. We need to have elements such as QR code detection running in the background. We compete in timed rounds and our background processes need to be saved to a file and submitted to the judges. Having a Save option for the log will expedite our process for capturing the log file for submission.

kwagyeman, 2019-03-15:
Okay, would you also like a record to file option where the data is written to a log file? Along with a copy and paste method?

t1jump, 2019-03-15:
Currently we can copy the data in the Serial Terminal by highlighting, then paste to Notepad/other. A direct Record to File would save the highlight/copy step. Record to File would also be more direct than a Save button (which would still be either a copy of the Serial Terminal data that would need to be pasted into Notepad, or require a request of file destination. A Record to File seems a better approach, and if it is easier to add to the IDE an even better option.

kwagyeman, 2019-04-09:
Hi, I've added this feature as just a save button. It will remember the file path where you were saving to. Recording is harder to do right as the IDE will need a start/stop button and it can get tricky then.

closed 2019-04-09 by kwagyeman