openmv/openmv-ide #38: Error while setting up IDE

rudrathegreat, 2019-03-17:
### Steps to reproduce
1.Use the docs, link below to install and setup the ide - 

https://docs.openmv.io/openmvcam/tutorial/software_setup.html

### Expected behaviour
When you run `./openmvide` or `./openmvide.sh` it should run the application.

### Actual behaviour
Comes up with the following error - 

```
./openmvide: error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory
```

### Describe your setup

**IDE version: 2.1.0**

**Firmware version:**

**Operating system:Raspbian**

### For developers

**GCC version:Dunno**

**Build target:Dunno**

kwagyeman, 2019-03-17:
I'll see what I can do about this... I guess I can install the last RPi build and see if that's the issue. If it's not then it's how your system gets setup.

kwagyeman, 2019-04-16:
Hi, OpenMV IDE runs great on a newly installed Pi that's totally clean. That's about the best I can debug this... if it's not running on your Pi it means you installed something else to break the IDE.

Did you run the setup.sh script first?

? https://www.raspberrypi.org/forums/viewtopic.php?t=191857

closed 2019-04-16 by kwagyeman

josetvrs, 2022-01-27:
I'm having the same problem. Could you solve it?