openmv/openmv-ide #39: make.py feature

bogflap, 2019-03-25:
At present make.py for linux (Opensuse in fact) searches for a directory Qt in the home directory.
Unfortunately I have a directory called QtCreatorWorkspace in my home directory which make.py finds as qtdir.
I am not that familiar enough with python to suggest a fix but perhaps searching for directory "Qt" only might help

kwagyeman, 2019-03-25:
Here?

https://github.com/openmv/openmv-ide/blob/master/make.py#L84

bogflap, 2019-03-26:
Yes, line 104

kwagyeman, 2019-03-26:
Just change the "Qt" string to "^Qt$" here:

https://github.com/openmv/openmv-ide/blob/master/make.py#L44

and 

https://github.com/openmv/openmv-ide/blob/master/make.py#L104

This forces the match to be on a string that just says Qt versus the first characters of it.

bogflap, 2019-03-27:
Okay

closed 2019-03-27 by bogflap