openmv/openmv-ide #43: Translation Problems

Hofer-Julian, 2019-05-02:
When I first connected our openmv cam I got a cryptical message. Only after I changed the language from German to English I saw that I should execute "sudo adduser...". Maybe the code could be changed so that this line does not get translated?

kwagyeman, 2019-05-02:
Hi, it must have been translated by our bulk translation service. Could you post the message here? I'll get it fixed in the next IDE release.

Hofer-Julian, 2019-05-02:
@kwagyeman not sure if I can reproduce it.

I probably would need to remove to user from the group again, right?

kwagyeman, 2019-05-03:
German translations are here if you want to search for it.

https://github.com/openmv/qt-creator/blob/b587ed9f6c93c6844782dc2e54c307878419c8e1/share/qtcreator/translations/qtcreator_de.po

Hofer-Julian, 2019-05-03:
It is this part:

```
#: ../../../src/plugins/openmv/openmvplugin.cpp:1808
msgctxt "OpenMV::Internal::OpenMVPlugin|"
msgid ""
"\n"
"\n"
"Try doing:\n"
"\n"
"sudo adduser %L1 dialout\n"
"\n"
"...in a terminal and then restart your computer."
msgstr ""
"\n"
"\n"
"Versuchen Sie Folgendes:\n"
"\n"
"Übernehmen Sie User %L1 Dialout\n"
"\n"
" ... in einem Terminal und starten Sie Ihren Computer neu."
```

"Übernehmen Sie User %L1 Dialout\n" should be changed  to ""sudo adduser %L1 dialout\n". Maybe you could make it so that this problem doesn't occur for other languages as well

kwagyeman, 2019-05-03:
I think the beat I can do it to make sure the command text is not in the translation string. Then Google won't translate it.

Hofer-Julian, 2019-05-03:
@kwagyeman that's what I wanted to suggest :)

kwagyeman, 2020-01-20:
Fixed

closed 2020-01-20 by kwagyeman