openmv/openmv-ide #47: IDE file copy issue

nezra, 2019-06-06:
### Steps to reproduc
1. create a custom file you'll be importing. a complete script isn't necessary.
2. use "#%" in the file line, such as this: 

print("\tSending param #%d is %d bytes long" % (i, len(param)))

3. Import a custom file into the script 
4. start the script, and you are presented with the option for openmv to upload the file to the device. 

### Expected behaviour
library is uploaded to the device, unaltered.

### Actual behaviour
because the IDE strips all of the comments and extra spaces out, it shortens the line above to:

print("\tSending param 

which breaks the rest of the code. Since the IDE attempts to version check it both ways, any time an included file is different from one or the other device, you either have to do it manually by copying/pasting both ways before starting the script.

### Describe your setup

**IDE version:**
2.2.0

**Firmware version:**
3.4.1

**Operating system:**
Windows 10

### For developers

**GCC version:**

**Build target:**

kwagyeman, 2020-01-20:
Going to work on fixing this now. We no longer support libraries that the IDE has to copy from our code anymore so maybe I can handle modules in folders. Let me see...

kwagyeman, 2020-02-01:
This is too complex for our next release to fix. I will work on it later. This will get fixed however. I will be doing a lot of work on the IDE starting in April.

kwagyeman, 2022-05-14:
Comment stripping was fixed. However, the library copying needs to be added.

kwagyeman, 2023-05-04:
@nezra I'd like to close out this issue and it's been a long time.

The code for this has been update a few times and I don't even know if this is an issue anymore. I just put a fix into the code base for this issue... however, I don't know if it resolves the error you were having. Can you post the files you had an issue with? I can get this fixed right away now.

kwagyeman, 2023-05-04:
Okay, the IDE definitely handles copying over modules that have submodules in them... since it just treats all that as one giant module.

As for the a single python file. Let me see if I can prevent it from flattening the directory structure.

kwagyeman, 2023-05-04:
Okay, working on this. Figured out what to do. Should be able to have the IDE mirror the folder structure in the Documents folder on demand.

kwagyeman, 2023-05-04:
Okay, this is fixed. The IDE will correctly update modules and maintain a folder structure.

closed 2023-05-04 by kwagyeman