openmv/openmv-ide #48: make: *** No rule to make target 'installer'.  Stop.

jiapei100, 2019-06-09:
I'm trying to build openmv-ide from source. However, ```./make.py``` brings me the following **ERROR** message:

```
./make.py
......
make[1]: Leaving directory '.../openmv/openmv-ide/build/bin'
make: Leaving directory '.../openmv/openmv-ide/build'
make: *** No rule to make target 'installer'.  Stop.
Make Failed...
```

Why is it so?

kwagyeman, 2019-06-09:
Dunno. I don't get that issue. This would imply the build script is calling make with the wrong argument.

jiapei100, 2019-06-09:
@kwagyeman 
I didn't add any arguments so far.
Just trying to build everything from scratch ...

Do I need to set a value for variable **installer** in file **make.py** ?
BTW, I'm usling **Python 3.7.3** .

kwagyeman, 2019-06-10:
I don't really know unless you provide more context about what platform you are using.

I'm guessing you don't have the Qt Installer framework installed. E.g. this line:

https://github.com/openmv/openmv-ide/blob/master/make.py#L228

The make.py script has to find the Qt installer framework on the system.

jiapei100, 2019-07-06:
@kwagyeman 

I noticed that in my **Makefile**
```
....../openmv-ide/build/Makefile
```
there is **NO** target **installer** at all, but **ONLY** :

* bindist
* bindist-installer
* bindist_debug
* install_appstream
* install_desktop
...

This is so weird.

jiapei100, 2019-07-06:
I guess there are some bugs in **make.py** .

I clearly see **line 638**  in file **openmv-ide/build/Makefile** :
```
install:install_subtargets install_appstream install_desktop  FORCE
```
But, there is NO **installer** as the target.

However, where is **openmv-ide-*.run** ?

```
installer = glob.glob(os.path.join(builddir, "openmv-ide-*.run"))[0]
IndexError: list index out of range
```

jiapei100, 2019-07-06:
@kwagyeman 

I got it:
I'm using the **NEWEST** qt-creator, which does NOT build **installer**.

kwagyeman, 2020-01-20:
Closing this issue. I just install qtcreator and then the qtinstaller framework and it builds. These are separate things. I have like a 3 year old version of things on my PC. While I understand it might not build with the latest tools... this isn't something that I operationally have time to fix for others.

closed 2020-01-20 by kwagyeman