openmv/openmv-ide #56: implement sync in Linux

carterw, 2019-12-20:
Running the IDE on Ubuntu Linux.
The 'save open script to openmv cam' action currently requires that a 'sync' command be executed in a separate window, otherwise the file write will usually remain cached and not be flushed out to the hardware. The sync could be performed automatically by the IDE after the file copy (which actually seems to be a delete followed by a write?).

If it can be done as a system call from within code I think it is fsync().

kwagyeman, 2020-01-20:
The IDE does call sync. Linux ignores it. Let me see what I can do... Maybe there's something else I can call.

iabdalkader, 2020-01-31:
Would be really great if you fix this next.

kwagyeman, 2020-02-01:
Added sync command for linux.

closed 2020-02-01 by kwagyeman