openmv/openmv-ide #67: How to package and publish openmvide

Kevincoooool, 2020-02-19:
I have compiled openmvide using QT and it can run normally, but it can only be opened and run in QT. After referring to the readme.md operation under win10 environment, the prompt:

PS D: \ GitHub \ openmv-ide> python. \ Make.py
Traceback (most recent call last):
   File ". \ Make.py", line 266, in <module>
     make ()
   File ". \ Make.py", line 202, in make
     "&& jom installer INSTALL_ROOT =" + installdir + "IFW_PATH =" + ifdir):
TypeError: cannot concatenate 'str' and 'NoneType' objects

Is this a syntax error?
I have installed 7-zip python2.7 QT5.8 QT installer freamwork normally

kwagyeman, 2020-02-19:
It's not an error in the script. The variable IFW_PATH is invalid as the script can't find the installer framework. Please check the path you installed it to matches where the script looks.

closed 2020-02-19 by kwagyeman

Kevincoooool, 2020-02-20:
> It's not an error in the script. The variable IFW_PATH is invalid as the script can't find the installer framework. Please check the path you installed it to matches where the script looks.

Hello, I re-installed again, the path is all correct, but it still prompts an error：
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:57:20: note: 'class QTransform' declared here
 class Q_GUI_EXPORT QTransform
                    ^~~~~~~~~~
In file included from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qbrush.h:50,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qpalette.h:46,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtWidgets/qwidget.h:48,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtWidgets/qframe.h:44,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include\QtWidgets/qlabel.h:44,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include\QtWidgets/QLabel:1,
                 from ..\..\..\..\qt-creator\src\libs\utils\statuslabel.h:33,
                 from ..\..\..\..\qt-creator\src\libs\utils\statuslabel.cpp:26:
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h: In member function 'QTransform& QTransform::operator=(QTransform&&)':
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:81:46: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class QTransform' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
     { memcpy(this, &other, sizeof(QTransform)); return *this; }
                                              ^
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:57:20: note: 'class QTransform' declared here
 class Q_GUI_EXPORT QTransform
                    ^~~~~~~~~~
In file included from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qbrush.h:50,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qpen.h:45,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include\QtGui/qtextformat.h:49,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include\QtGui/QTextCharFormat:1,
                 from ..\..\..\..\qt-creator\src\libs\utils\ansiescapecodehandler.h:31,
                 from ..\..\..\..\qt-creator\src\libs\utils\outputformatter.cpp:26:
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h: In member function 'QTransform& QTransform::operator=(QTransform&&)':
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:81:46: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class QTransform' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
     { memcpy(this, &other, sizeof(QTransform)); return *this; }
                                              ^
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:57:20: note: 'class QTransform' declared here
 class Q_GUI_EXPORT QTransform
                    ^~~~~~~~~~
In file included from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qimage.h:48,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qpixmap.h:49,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qicon.h:46,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include\QtWidgets/qstyle.h:47,
                 from D:\Qt\Qt5.8.0\5.8\mingw53_32\include\QtWidgets/QStyle:1,
                 from ..\..\..\..\qt-creator\src\libs\utils\flowlayout.h:32,
                 from ..\..\..\..\qt-creator\src\libs\utils\flowlayout.cpp:26:
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h: In member function 'QTransform& QTransform::operator=(QTransform&&)':
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:81:46: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class QTransform' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
     { memcpy(this, &other, sizeof(QTransform)); return *this; }
                                              ^
D:\Qt\Qt5.8.0\5.8\mingw53_32\include/QtGui/qtransform.h:57:20: note: 'class QTransform' declared here
 class Q_GUI_EXPORT QTransform
                    ^~~~~~~~~~
jom: D:\GitHub\openmv-ide\build\src\libs\utils\Makefile [release] Error 2
jom: D:\GitHub\openmv-ide\build\src\libs\Makefile [sub-utils-make_first] Error 2
jom: D:\GitHub\openmv-ide\build\src\Makefile [sub-libs-make_first-ordered] Error 2
jom: D:\GitHub\openmv-ide\build\Makefile [sub-src-make_first-ordered] Error 2
Make Failed...

kwagyeman, 2020-02-20:
Looks like some type of library mismatch with the code. Note that I'm using a very old Qt version to compile the IDE from about 3 years ago. Haven't updated it yet because I haven't wanted to fight issues like the above.