openmv/openmv-ide #71: Installing IDE on Rapsberry 4

matiasandina, 2020-03-31:
I had problems related to the ones discussed [in the forum](https://forums.openmv.io/viewtopic.php?t=1515)

I solved it by direct `apt-get install` of conflicting libraries. IDE works even though I have no monitor plugged in and doing remote-control through VNC.

```
sudo apt-get install libgles2-mesa-dev
sudo apt-get install libpng-dev
sudo apt-get install libpng12-dev
sudo apt-get install qt5-default
sudo apt-get install libts-dev
```

Maybe adding these to `setup.sh` can help others ?

matiasandina, 2020-03-31:
Also, at the expense of being annoying, could you implement a desktop launcher during setup.sh?

This is how my `openmv.desktop` inside `/home/pi/Desktop` looks like. And I just downloaded your `.jpg` logo and stored it on the install folder.

```
[Desktop Entry]
Type=Application
Comment=openmvIDE
Name=openmvIDE
Exec=/home/pi/openmvide/bin/openmvide.sh
Icon=/home/pi/openmvide/openmv-logo.jpg
Terminal=false
Categories=Utility
StartupNotify=true
```

kwagyeman, 2020-03-31:
Sure, good idea. We will have a new IDE release in 1.5 weeks.

matiasandina, 2020-03-31:
oh no! I just finished installing 10 Raspberries! :sob:

kwagyeman, 2020-06-21:
Done

closed 2020-06-21 by kwagyeman