openmv/openmv-ide #72: OpenMV IDE crashes with SEGFAULT on start

vatula, 2020-04-24:
Fedora 32, python dependencies installed
running ./openmvide or ./openmvide.sh either with sudo or rootless results in
`Segmentation fault (core dumped)`
journalctl entry is not verbose. This is what it has:
```
Apr 24 20:11:39 attic systemd[1]: systemd-coredump@13-224743-0.service: Succeeded.
-- Subject: Unit succeeded
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- The unit systemd-coredump@13-224743-0.service has successfully entered the 'dead' state.
Apr 24 20:11:39 attic audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@13-224>
Apr 24 20:11:39 attic audit: BPF prog-id=272 op=UNLOAD
Apr 24 20:11:39 attic audit: BPF prog-id=271 op=UNLOAD
Apr 24 20:11:39 attic audit: BPF prog-id=270 op=UNLOAD
Apr 24 20:11:39 attic abrt-server[224760]: Executable '/home/atticdweller/openmvide/bin/openmvide' doesn't belong to any package and ProcessUnpackaged is >
Apr 24 20:11:39 attic abrt-server[224760]: 'post-create' on '/var/spool/abrt/ccpp-2020-04-24-20:11:39.367314-224688' exited with 1
Apr 24 20:11:39 attic abrt-server[224760]: Deleting problem directory '/var/spool/abrt/ccpp-2020-04-24-20:11:39.367314-224688'
...skipping...
                                                
                                                Stack trace of thread 224688:
                                                #0  0x00007fb717659fb1 n/a (/home/atticdweller/openmvide/lib/Qt/lib/libQt5Network.so.5.7.0 + 0x102fb1)
                                                #1  0x0000000000000201 n/a (n/a + 0x0)
-- Subject: Process 224688 (openmvide) dumped core
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: man:core(5)
-- 
-- Process 224688 (openmvide) crashed and dumped core.
-- 
-- This usually indicates a programming error in the crashing program and
-- should be reported to its vendor as a bug.
```

kwagyeman, 2020-04-24:
I don't know if the IDE runs on Fedora.

Typically you'd get a missing dependency error and it wouldn't run. Getting a segfault is something else.

You'll have to debug what's going on if you want it to run on Fedora 32.

Probably if you compile form source it will run on that system.

kwagyeman, 2023-07-08:
Closing this as outdated once the new IDE version comes out.

closed 2023-07-08 by kwagyeman