openmv/openmv-ide #74: OpenMV IDE not displaying the example folders contents.

funtimetinker, 2020-04-26:
### Steps to reproduce
1. Open OpenMV IDE.
2. Navigate to Help menu bar item, type search term in search box.
3. Navigate to File menu bar item, navigate to example folders.

### Expected behaviour
Examples should display folder contents.

### Actual behaviour
Folder contents are not displayed, after multiple searches

**IDE version:**
OpenMV IDE 2.4.0
[image]

**Operating system:**
macOS 10.15.4

NoRez4U, 2022-11-18:
Seeing same/similar bug on Windows 10 Pro, 21H2. For me, *sometimes* no drop down appears when clicking on the MENU items. Click on the menu item once and it highlights, click on it again and it deselects, but nothing drops down. Sometimes it happens on only one menu item while other times it affects multiple menu items. I'll watch it more b/c I think there is a trigger causing it. 

Currently using OpenMV IDE 2.9.7

kwagyeman, 2022-11-18:
I have to update the Qt version to fix this.

kwagyeman, 2023-07-08:
Hi, this is fixed in the upcoming version of the IDE. The new Qt system resolves the issue.

closed 2023-07-08 by kwagyeman