openmv/openmv-ide #82: arm64 build fails

jimmont, 2020-07-10:
In trying to build for an Raspberry Pi 4 running Ubuntu 20.04LTS 64-bit server based on the instructions failed with the following.
Is it possible to provide a compatible IDE or updated build instructions for Ubuntu 20.04 LTS 64-bit? It appears likely this would work with 64-bit Raspberry Pi OS once it is out of beta.

Thank you.

```git remote -v
origin	https://github.com/openmv/openmv-ide.git (fetch)
origin	https://github.com/openmv/openmv-ide.git (push)

git status
On branch master
Your branch is up to date with 'origin/master'.

apt install build-essential libgl1-mesa-dev
apt install qtcreator qt5-default qtdeclarative5-dev libqt5serialport5-dev

Qt Creator 4.11.0
based on Qt 5.12.8 (GCC 9.3.0, 64bit)

cat /etc/lsb-release                                                          
DISTRIB_ID=Ubuntu                                                             
DISTRIB_RELEASE=20.04                                                         
DISTRIB_CODENAME=focal                                                        
DISTRIB_DESCRIPTION="Ubuntu 20.04 LTS"                                        

uname -a                                                                      
Linux ubuntu 5.4.0-1013-raspi #13-Ubuntu SMP Mon Jun 15 03:17:37 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux

cd openmv-ide
./make.py --rpi &> build.log   
```
[build.log](https://github.com/openmv/openmv-ide/files/4901933/build.log)

kwagyeman, 2020-07-10:
You have to fix the build errors yourself. If you can get it building then please send a PR for the fixes in.

kwagyeman, 2020-07-10:
Note, I don't build the IDE on the RPI. I build it on a VM running Ubuntu with a cross's compiler using the qtrpi project.

closed 2023-07-08 by kwagyeman