openmv/openmv-ide #85: duf-util is not executable in Mac OSX binary, causing bootloader install to fail

tstellanova, 2020-08-05:
When attempting to install bootloader on OpenMV H7 board using Mac OSX, the `dfu-util` binaries are not executable, which causes the bootloader install to fail.  The workaround is:

```
cd Applications/OpenMV\ IDE.app/Contents/Resources/dfu-util/osx/
 chmod +x dfu-util
 chmod +x dfu-suffix
 chmod +x dfu-prefix
```

Shell error trace (when dfu-util fails):

```
Last login: Tue Aug  4 18:16:31 on ttys006
/var/folders/7n/hvxxf92n2czg2lv862k0v5bm0000gn/T/openmvide-dfu-util.sh ; exit;
todd@skylab ~ % /var/folders/7n/hvxxf92n2czg2lv862k0v5bm0000gn/T/openmvide-dfu-util.sh ; exit;
/var/folders/7n/hvxxf92n2czg2lv862k0v5bm0000gn/T/openmvide-dfu-util.sh: line 3: /Applications/OpenMV IDE.app/Contents/Resources/dfu-util/osx/dfu-util: Permission denied

[Process completed]
``

kwagyeman, 2020-08-05:
Ah, thanks for catching this. Must have happened because the binaries were extracted on a PC.

kwagyeman, 2020-10-18:
Fixed

closed 2020-10-18 by kwagyeman