openmv/openmv-ide #86: Can't click on any menu items

frank26080115, 2020-08-08:
Windows 10 Pro 64 bit, Version 10.0.19041 Build 19041

IDE version is 2.6.0

It works 99% of the time but something would happen, I don't know what causes it, but the IDE stopped letting me click on any menus. Please see attached video in uploaded zip file
[openmvide_cantclickbuttons.zip](https://github.com/openmv/openmv-ide/files/5044945/openmvide_cantclickbuttons.zip)

Closing the IDE is the only solution that I found

kwagyeman, 2020-08-08:
This is a known issue. It's related to qtcreator actually and the base ide I built off of. I don't know what causes it. I think it's a memory corruption however.

frank26080115, 2020-08-10:
err I keep getting bitten by this bug

I get that memory corruption bugs are like impossible to hunt down...

Would you consider actually having tabs? Also, using CTRL-TAB to cycle through tabs? It would help me keep working since I can still use CTRL-S to save.

And why can't I type into the serial terminal? Is that a bug? I have like a billion new files just to test like 5 lines of code that I should be just running in the terminal. Swapping between RealTerm and the IDE is a pain in the ass too so I'm not going to do that.

kwagyeman, 2020-08-10:
There's a plugin that I can add to Qtcreator to get tabs. It's like, alot of work. So, I have no plans to do that right now.

Use the Tools->Open Terminal Feature to get a terminal you can type in. The default terminal is just a debug log since we don't have debug message support for serial input into the camera in debug mode. It's not a bug so much as it's not implemented at all. I can turn on typing in the IDE easily. But, a link to the camera to send data needs to be added. @iabdalkader. Can you think about adding a debug message path for this? Arudino may want it.

kwagyeman, 2021-01-30:
I found what causes this. Actually, the menu appears behind the IDE window. It's actually a bug in Qt. There's some literature online where the issue gets fixed in the very latest releases. I'll look into getting this fixed as it bothers me too. However, it's somewhat of a challenge to upgrade our IDE code base to the very latest. I'm not sure if I can work around without rebuilding Qt itself.

YCSamSun, 2022-04-25:
I still suffer this proble on ubuntu 20.04 version. Is there any solution for this?

kwagyeman, 2022-04-25:
I have to update the IDE qt backend. It's not fixable without a huge upgrade and spend weeks on it which I don't have right now.

kwagyeman, 2023-07-08:
Hi, this should be fixed with the latest IDE about to be released.

closed 2023-07-08 by kwagyeman