openmv/openmv-ide #92: Main Serial Terminal doesn't send key-presses to camera, Tools / Open Terminal does

JamesNewton, 2020-10-13:
Using the USB_VCP import from pyb and then 
`usb = USB_VCP()`
with a 
`cmd = usb.recv(1, timeout=10)`
and a series of e.g. 
`if (cmd == b'!'):`
is a really nice way to allow a single script in the camera to serve multiple functions or to have adjustable parameters. It makes it much more useful, at least it has for our applications:
https://github.com/HaddingtonDynamics/Dexter/issues/107

One thing that is a bit sad is that when we are using the IDE, we can't use the main Serial Terminal to send characters to the camera. We can from a serial terminal (e.g. PuTTY or screen) and we can after opening a secondary terminal window via the Tools / Open terminal menu. In that mode, however, we don't get to see what the camera is seeing: the "Frame Buffer" window is empty and can not be enabled. 

Suggestions:
- If the user tries to type into the main Serial Terminal, indicate that this isn't supported. It took us a LONG time to realize we had working code in the camera and the IDE just wasn't sending anything through. Pointing people to the Tools menu for that would be a great help. e.g. "Sending data not supported from main terminal window. Use Tools / Open terminal for 2 way serial communication."
- In the Tools / Open terminal, indicate with a message in the Frame Buffer that this isn't supported and send people back to the main IDE if they want to see the image. e.g. "Supported only in Main IDE"
- Obviously, if possible and as time allows, just support sending data from the main serial terminal and support the frame buffer in the serial terminal.

kwagyeman, 2020-10-13:
Hi,

You can send images to the non-main terminals.

Do: 

print(img.compressed_for_ide())

That's it.

This is in the API. I literally made a video on this a while back. Years ago.

...

Sending data in the main terminal is not supported right now since there's no sub in the openmv protocol for it. Bug @iabdalkader to implement it and then I can make this work easily.

JamesNewton, 2020-10-13:
So I would suggest putting a message in the non-main terminal video window:
"Use `print(img.compressed_for_ide())` to send images here"
One line of doc where users need it beats entire videos that they don't know about. 

Maybe also link to that video from 4.10 in the doc
https://docs.openmv.io/openmvcam/tutorial/openmvide_overview.html?highlight=tools%20open%20serial%20terminal#open-terminal

kwagyeman, 2020-10-13:
That would be a good idea. 👍

JamesNewton, 2020-10-14:
If I could find the source code for the IDE, I'd add that message myself, but your repo confuses the heck out of me. I can't understand where the actual code is. LOL. Sorry.

davidhbrown, 2021-06-04:
Having mildly embarrassed myself making this error, too, I thought I'd see whether the IDE's tab could more simply be renamed "Serial OUTPUT" instead of the more confusing "Serial Terminal"... but like JamesNewton, I can't find code in here... I think the text is in the qt-creator fork (different repo linked to this one), specifically in share/qtcreator/translations/ :  https://github.com/openmv/qt-creator/tree/openmv/share/qtcreator/translations   Will have to spin up a "real" VM, though... couldn't install qt5-default under Windows Subystem for Linux's Ubuntu

kwagyeman, 2021-06-04:
We're going to add the feature eventually.

kwagyeman, 2022-05-14:
The IDE will send characters in the next release.

closed 2022-05-14 by kwagyeman

philputman, 2024-02-03:
Is this true? I couldn't get input() to work in version 4.1.5.

kwagyeman, 2024-02-03:
It's been implemented in the IDE but not in the firmware to do anything with the characters: https://github.com/openmv/openmv/issues/1151

https://github.com/openmv/openmv/blob/master/src/omv/common/usbdbg.c#L343

So, they do make it to the camera. I just need to pipe those into something. Same for setting the RTC.