openmv/openmv-ide #95: Installer won't update existing installation

frank26080115, 2020-10-21:
Windows 10, upgrading IDE to v2.6.5

The installer won't let me upgrade over an existing installation of OpenMV IDE

kwagyeman, 2020-10-21:
Delete the current installation and then install the new one. The installer frame work we use from at won’t let us update in place.

frank26080115, 2020-10-21:
maybe use fileExists() and setUpdater() in the script? or offer to call the uninstaller if fileExists() returns true?

kwagyeman, 2020-10-21:
I think in later versions of the tool they fixed this. I just haven’t made it a serious priority to fix.

I’ll leave it as a bug, then take a look for the next release.

falthaus, 2022-02-28:
On Win10, upgrading from IDE v2.8.1 to v2.9.0, the old version again has to be deinstalled manually first (same behaviour on two different Win10 machines).
It's a while since v2.8.1 came out, but I thought this used to be fixed(?)

kwagyeman, 2022-02-28:
Yeah, I haven't fixed it yet.

We will be fixing it in the next release. I will be upgrading the Qt Version and Qt Installer framework version. This will also result in a loss of support for 32-bit systems. Newer versions of Qt only support 64-bit.

kwagyeman, 2022-04-25:
Fixed

closed 2022-04-25 by kwagyeman