openmv/openmv-ide #97: Simplify DFU firmware installation dialog

sebromero, 2020-11-02:
The current version of the dialog window that appears when connecting a board in DFU mode is a tad information heavy. I'd suggest to focus on the primary call to action and "hide" the "advanced" options in a dropdown menu. -> See mockup below

Also I don't fully understand the difference between "Erase Onboard Data Flash" and "Erase the internal file system?". Is it the same? In any case, I'd suggest making this an option of the dialog window and adding a tooltip or a label explaining what this option does.

[image]
[image]

**Mockup**
[image]

kwagyeman, 2020-11-03:
Hi, I can partially do this easily. Please understand that the way to program the board is through the bootloader. Because we have to support Arduino's different DFU bootloader this is unnecessarily compilated.

I'll simply this menu and add some help some more text strings.

kwagyeman, 2020-11-03:
Fixed in next release.

closed 2020-11-03 by kwagyeman