openmv/openmv-ide #98: Add Dark Mode

kwagyeman, 2020-11-02:
Add option to switch between light and dark mode.

baljo, 2021-02-03:
Hi, any ETA on this one? If implemented in next release, I'd buy you a cup of coffee/tea :-)
BTW, I am the one asking for it in the forums.

kwagyeman, 2021-02-03:
It's like very hard. Reskinning the whole IDE is not trivial.

kwagyeman, 2021-02-03:
The ETA for this is pretty much infinite without external help.

dewhisna, 2021-02-03:
Actually, it might be more trivial than you think.  Being Qt Creator based, you should be able to just load a new style sheet when you launch it, which is just a command-line option. Here's one that's available: [https://draculatheme.com/qtcreator/](https://draculatheme.com/qtcreator/)

kwagyeman, 2021-02-03:
Sorry, the current mode is dark mode. What you guys want is light mode.

A lot of the widgets are custom styled by me. They aren't done so through QtCreator. So, I'll have to add this option and then fix styling for one widget after another. This is probably 20+ hours of work.

As I have to focus on drivers like triple buffering for the camera I don't really have time.

kwagyeman, 2021-02-03:
If you want it done convince Arduino folks it needs to be done. Then I can get contracting hours for it.

baljo, 2021-02-04:
Hmm, a pity, I believe there are as many who loves dark modes as who cannot stand them for different reasons. My reason are simple practical, not based on feelings. I can't use dark mode at all as I need to turn up the brightness too much which kills my eyes, latest when switching to another application.

kwagyeman, 2021-02-04:
No, I understand dark mode has issues. In particular, it's quite a challenge to use outside in sunlight.

Okay, so, this is easy to do... I can allow the editor style to change to a different mode. This will look uglyish, but, change 70% of the application to light mode.

baljo, 2021-02-04:
> No, I understand dark mode has issues. In particular, it's quite a challenge to use outside in sunlight.
> 
> Okay, so, this is easy to do... I can allow the editor style to change to a different mode. This will look uglyish, but, change 70% of the application to light mode.

Awesome! If needed, I'd be happy to test it when (not if :-) ) you have it ready and give feedback!

baljo, 2021-05-27:
Any hope to have option of light mode in next IDE-version?

kwagyeman, 2021-05-27:
Nope, however, tabs, split view, and minimap in the scroll bar will come.

For light mode to happen I honestly need help.

baljo, 2021-05-28:
Ok, unfortunately I don't have the knowledge or capacity to help myself. Am though willing to test when the time comes.

baljo, 2023-05-15:
Now when you're upgrading Qt libraries, any chance of getting a light mode option?

kwagyeman, 2023-05-15:
Yeah, Qt 6.5 which will be updating to has this feature. So, it will be across the board.

baljo, 2023-05-17:
Great news, at least for me!

kwagyeman, 2023-06-15:
This is fixed for version 4.0.0. Light mode will also land too. The whole IDE is themed now so it can be whatever color you like.

closed 2023-06-15 by kwagyeman