openmv/openmv-ide #104: IDE copy & paste removes tabs

iabdalkader, 2021-01-18:
IDE copy & paste code the indentation is lost, fortunately you can select a block and hit tab, but still it should be fixed.

kwagyeman, 2021-01-31:
It actually does work... if you paste on a line with text right above it keeps the indentation. However, if there's white space then it doesn't. Maybe I can fix that.

kwagyeman, 2021-01-31:
`// Indent a text block based on previous line.
// Simple text paragraph layout:
// aaaa aaaa
//
//   bbb bb
//   bbb bb
//
//  - list
//    list line2
//
//  - listn
//
// ccc
//
// @todo{Add formatting to wrap paragraphs. This requires some
// hoops as the current indentation routines are not prepared
// for additional block being inserted. It might be possible
// to do in 2 steps (indenting/wrapping)}
//

using namespace TextEditor;

int NormalIndenter::indentFor(const QTextBlock &block, const TabSettings &tabSettings)
{
    Q_UNUSED(tabSettings);

    QTextBlock previous = block.previous();
    if (!previous.isValid())
        return 0;

    const QString previousText = previous.text();
    // Empty line indicates a start of a new paragraph. Leave as is.
    if (previousText.isEmpty() || previousText.trimmed().isEmpty())
        return 0;

    return tabSettings.indentationColumn(previousText);
}`

kwagyeman, 2021-01-31:
Fixed it. This was actually correct behavior in the IDE. It did not consider empty lines of only white space indentation.

closed 2021-01-31 by kwagyeman