openmv/openmv-ide #105: Improve DFU window output.

iabdalkader, 2021-01-20:
[image]
Please check if the line ends with "bytes\r" or "bytes\n" and if so, move the cursor to 0 instead of printing a new line, it will look much better.

kwagyeman, 2021-01-30:
This is really hard to fix. If Arduino is paying for me to do then I will. The text arrives in an unbuffered manner. So, I will have to make a statemachine to parse it and determine where line ends are and etc.

iabdalkader, 2021-01-30:
I requested this because it will look much better. You only need to check for line ends and when you get a line end, scroll back to the beginning of the line, or if you keep a buffer for the line overwrite it. Points 2 and 3 are optional.

kwagyeman, 2021-01-30:
Yeah, this is just a lot more involved that you think on the way the qplaintext edit works.

kwagyeman, 2021-08-01:
This looks fine on apps line picotool. So, it's an issue with dfu-util. Won't fix.

iabdalkader, 2021-08-02:
Same issue with picotool and bossac

kwagyeman, 2022-05-02:
Fixed

closed 2022-05-02 by kwagyeman