openmv/openmv-ide #107: OpenMV windows installation issue

bsyoo77, 2021-01-27:
Hi, I got this error message below as I try to install the OpenMV IDE (windows 2.6.8). Can you please help me to resolve this issue?
-----------------------------------------------------------------------
[image]

Preparing the installation...

Installing component OpenMV IDE Application...
Extracting "2.6.8openmv-ide-windows-2.6.8-installer-archive.7z"
Done
Writing maintenance tool.

Installation aborted!

kwagyeman, 2021-01-27:
Please uninstall the IDE totally and then install again.

bsyoo77, 2021-01-28:
I could not install the OpenMV, so Uninstall cannot be executed. After the installation failure, the attached folder was created in the C:\Program Files (x86) folder. Please let me know if there is an alternative way to install the IDE.
[OpenMV IDE.zip](https://github.com/openmv/openmv-ide/files/5883866/OpenMV.IDE.zip)

bsyoo77, 2021-01-28:
The installation just froze at 66% and going back to 0% installation stage. This is the same issue found here: https://forums.openmv.io/viewtopic.php?t=699

kwagyeman, 2021-01-28:
Can you try running the installer as an admin? I don't know how to fix this issue. Typically, when this happens it's because of settings on your computer blocking an install.

bsyoo77, 2021-01-28:
Hi, I solved the problem. I am using Rygen 5 3600 with MSI Gaming Plus Max. This is a [common problem](https://forum.qt.io/topic/111473/maintenance-tool-error-cannot-open-file-for-writing-no-error/38) with AMD. After updating my BIOS (7/2019) with a recent one (11/2020) from MSI webpage, the OpenMV IDE installation worked. Thanks for your help.

kwagyeman, 2021-01-28:
Wow, how does window even work if this is happening. I feel like this would break so many parts of the operating system.

closed 2021-01-28 by kwagyeman