openmv/openmv-ide #109: Autoindent only works when pressing enter in the middle of a line, not at the end of a line

frank26080115, 2021-02-05:
IDE v 2.6.9, autoindent enabled

Type a line like

    ····testing·abc·123

hit enter, type "456", you get

    ····testing·abc·123
    456

expected result is

    ····testing·abc·123
    ····456

but the weird thing is, if you put the caret somewhere in the middle of the line, and hit enter, you get something like

    ····testing·ab
    ····c·123

which is what you'd expect from a text editor with autoindent

kwagyeman, 2021-02-05:
Thanks for pointing this out... weird. the behavior is very different from what was happening when I released the IDE on fixing this.

Like, I just fixed this issue and the behavior is very different. Than the fix I did...

kwagyeman, 2021-02-05:
Auto indent is now not working at all.

kwagyeman, 2021-02-05:
Uncheck enable automatic indentation in the options. This actually seems to do exactly what you don't want. I need to debug what's going on here.

The behavior then will be exactly what you want. I will leave this bug open so I can fix the dialog and what's going on behind the scenes.

kwagyeman, 2021-08-01:
Hide it... longer term fix for all behaviors later.

kwagyeman, 2023-06-22:
Figured out what was going on with this. It's now fixed.

closed 2023-06-22 by kwagyeman