openmv/openmv-ide #110: feature request: revert handling of missing file back to how it used to work

frank26080115, 2021-02-05:
The new v 2.6.9 IDE has started to immediately tell me when an open file no-longer exists. Which is pretty often since I'm unplugging the OpenMV all the time.

Previous versions of the IDE did not do any notifications at all, but upon attempting to save, it would claim it was read-only with a very elegant bar below the file name

The new version will force you to choose between closing the file or saving it somewhere else with a dialog

I much prefer the older way since I will be plugging the OpenMV back anyways later

I feel like editing files on the microSD card or internal flash would be a common enough use-case to justify my preference for many users. I know the IDE serves the currently-open file via REPL stdin, but I don't think it does any drive-mirroring or JIT mpy-cross compiling, so it makes sense for a user to have a ton of code files open in the IDE that are actually saved directly in flash or SD card, which makes sure that the latest code is in flash. This will be true for any project with more than one file. Forcing people to either close or move them whenever the OpenMV is unplugged is super annoying.

If you insist on notifying the user with a dialog box, try the way Notepad++ handles the same situation. The user has the option of just letting the file stay open.

If there's a difference between the file in flash vs in the editor when the OpenMV is re-plugged-in, then the editor usually asks if the user wants the copy on the device or the copy on the screen.

kwagyeman, 2021-02-05:
Hi, I didn't change the IDE logic on this. So, I don't know what's going on.

@iabdalkader - Did the way the camera writes data to the SD card change in the latest firmware? I think write. Maybe it's causing this?

iabdalkader, 2021-02-05:
The IDE has always been like that, this is 2.6.8

[image]

Nothing in the firmware could affect this.

iabdalkader, 2021-02-05:
Maybe add a checkbox "Don't show this message again"

frank26080115, 2021-02-05:
[image]

That's v 2.5.1

iabdalkader, 2021-02-05:
I see the issue, but I really don't think it has anything to do with the firmware... I think it's core qt creator functionality that may have changed with an update.

frank26080115, 2021-02-05:
This has nothing to do with firmware, sorry for the confusion but when I said flash, I meant the flash file system, not firmware.

kwagyeman, 2021-02-08:
What OS are you on? Linux/Windows has not changed here. Only Mac Qt was updated.

frank26080115, 2021-02-08:
Windows 10 thanks

kwagyeman, 2023-07-18:
Behavior seems to be back to what you wanted.

So, I will not do anything.

[image]

Note that the file syncing system now works in the IDE. So, you should edit code in your Documents folder. The IDE will automatically scan your import list when you run code and it will update the files on the camera if it sees the version on the camera are different that what's on the PC.

closed 2023-07-18 by kwagyeman