openmv/openmv-ide #111: Resolution and selected ROI different from snapshot

rongith, 2021-02-06:
IDE 2.6.9 / Windows 10

If the buffer has an image of W229 x H152 "sensor.set_windowing((196 , 82 , 229 , 152))":
Res displayed is 228x152
ROI selectable is 229x153
[image]

kwagyeman, 2021-02-08:
Noted. I'll see if I can fix the selectable window being 1-pixel too large.

rongith, 2021-02-08:
Plus, none of the three rectangles have the same dimension:
sensor.set_windowing : 229x152
Res: 228x152
ROI selectable: 229x153

kwagyeman, 2021-02-08:
Set windowing doesn't accept odd resolutions. So, it just removes 1 pixel. The ROI selectable in the IDE is just larger than should be allowed.

kwagyeman, 2021-05-28:
This is fixed in the latest firmware. Things are no longer rounded down.

closed 2021-05-28 by kwagyeman

rongith, 2021-05-28:
I'm not sure if is related as my bug is for IDE, but until the update I had som calculated ROI that may have been odd. Then sometimes I called sensor.set_windowing(roi) with some odd numbers.
I can't do that anymore with MT9V034 but only for the width. Else I have "ValueError: Sensor control failed!". You may update the documentation on the set_windowing function, I didn't find anything about it.

kwagyeman, 2021-05-28:
Hi, I’m the latest firmware the function is now extremely forgiving and does exactly what you want. It shouldn’t throw any errors even if you ask if to give you pixels out of bounds. It will just crop the rect.