openmv/openmv-ide #122: lost example scripts and documentation after update

kdarras, 2021-05-19:
I am running IDE 2.6.9 on Ubuntu 20.04 and since the last update a few days ago, I have the following:

- I lost example scripts (the .py files are still there but empty). error code: "cannot open example file "path...""
- I have no documentation displayed anymore when I hover the mouse on top of the purple functions in the code. When I click openMV docs, an empty index.html file appears

kwagyeman, 2021-05-19:
Please try deleting ./config/openmvide

This will remove settings and the IDE will have to rebuild them

kdarras, 2021-05-19:
I only have the .config/openMV folder and then OpenMVIDE.db and OpenMVIDE.ini files inside.

**Deleting the OpenMVIDE.ini file successfully triggered the download and rebuild.**

closed 2021-05-19 by kdarras