openmv/openmv-ide #126: The neural network file for number recognition cannot be found in OpenMV Ide2.6.9

D-Twelve-T, 2021-07-13:
Hello, I want to realize the task of identifying two digits of numbers, calculating the median, and sending the signal to STM32 microcontroller. I downloaded the IDE of 2.6.9, but I could not find the NN. Lenet neural network file in the CNN model network, and I could not find the relevant neural network file in the IDE folder either. May I ask how to solve this? Where can I find this neural network file to recognize numbers? I need him to tell me what the number is. Please help me,pelase. Thanks❗

kwagyeman, 2021-08-01:
Hi, this was removed. We switched to TensorFlow. You can create your own CNNs now.

closed 2021-08-01 by kwagyeman

Muddypond, 2021-11-04:
can we create our own CNN with openmv 4?

kwagyeman, 2021-11-04:
Yes