openmv/openmv-ide #127: Tabs need some improvments.

iabdalkader, 2021-08-02:
* The following common shortcuts/hotkeys should be used:

Ctrl + Page Up/Down -> Switch to the left or right tab.
Ctrl + Shit + Page Up/Down -> Move the tab itself to the left or right.
Ctrl + T -> open new tab (with new script)
Ctrl + W -> close the current tab.
Ctrl + Shift + T -> re-open the most recently closed tab

* The drop down list is not needed anymore, in fact it's confusing to have both. 

* Finally, the tabs should be on the same bar where the drop-down list is/used to be, and the new bar should be remove. For example:
[image]

kwagyeman, 2021-08-05:
Ctrl + Page Up/Down -> Switch to the left or right tab. (fixed)
Ctrl + Shit + Page Up/Down -> Move the tab itself to the left or right. (fixed)
Ctrl + T -> open new tab (with new script) -> use ctrl+N.
Ctrl + W -> close the current tab. (this is already implemented)
Ctrl + Shift + T -> re-open the most recently closed tab (qt-creator editor manager doesn't store this history)

The drop down list is not needed anymore, in fact it's confusing to have both. -> Punt to next-release/Won't fix
Finally, the tabs should be on the same bar where the drop-down list is/used to be, and the new bar should be remove. For example: -> Punt to next-release/Won't fix

iabdalkader, 2021-08-05:
> The drop down list is not needed anymore, in fact it's confusing to have both. -> Punt to next-release/Won't fix
> Finally, the tabs should be on the same bar where the drop-down list is/used to be, and the new bar should be remove. For example: -> Punt to next-release/Won't fix

Should keep this open then.

iabdalkader, 2021-08-21:
The Ctrl + Page Up/Down are swapped, page up should go to left tab, page down should go to right tab.

kwagyeman, 2022-02-17:
Fixed

closed 2022-02-17 by kwagyeman