openmv/openmv-ide #128: Add hotkey to reload firmware.

iabdalkader, 2021-08-02:
Would save a lot of time when testing small changes a lot, it should use the most recently used path and skip the "Tool->Run Bootloader->Run dialog" I suggest ctrl+shit+L

kwagyeman, 2021-08-05:
This is rather dangerous issue for users to accidentally activate. But, okay, not that hard. But, it still takes quite a bit of time to load firmware. I don't see the need for this. Will leave the issue open.

iabdalkader, 2021-08-05:
That's why it's a longer key combination (ctrl + shit + l), and it's useful because it saves the time needed to go through the bootloader dialog. These small things add up when you load fw 10s of times in a row.

iabdalkader, 2021-08-21:
Alternatively, the shortcut could show up the "run bootloader" dialog with run button having focus.

kwagyeman, 2022-02-23:
Ctrl + Shift + L -> launches Run Bootloader Dialog
Ctrl + Shift + E -> Launches Erase Flash Disk Dialog

closed 2022-02-23 by kwagyeman