openmv/openmv-ide #129: AltGr misbehaving on OpenMV IDE 2.8.1

fickludd, 2021-08-22:
I just upgraded to IDE 2.8.1 (tabs, yey). However this seems to have broken keyboard input. I’m on a Swedish keyboard layout, where {} brackets are on AltGr-7 to AltGr-0. Nothing happens when I try to input these. This is quite annoying, fix would be nice :).

falthaus, 2021-08-28:
Same here: IDE 2.8.1 on Windows 10 Home with a Swiss keyboard layout and on two different computers.
- [AltGr]+[0..9] get ignored, so do [Ctrl]+[Alt]+[0..9] (which should have the same effect)
- [AltGr]+[e] would work (on the Swiss keyboard that's the Euro sign - obviously not needed for python, but it seems some [AltGr] stuff seems to work)
- I only have this issue with OpenMV IDE, so I guess it's not on Windows this time.
- it doesn't matter what Language I set in Options->User Interface->Language

[AltGr]+[3] is the '#' sign, so commenting is a bit difficult now...

kwagyeman, 2021-08-29:
Hi, tab shortcuts are generated by:

QStringLiteral("Ctrl+Alt+%1");

Where %1 is replaced with 1 through 0.

What should I replace this with? I can remove the feature. Would like to keep it. But, I don't know if that's possible.

kwagyeman, 2021-08-29:
Also, what OS are you both using? I will cut a new release quickly with the fix.

fickludd, 2021-08-30:
I'm on Win 10 Pro. I would

  1) remove the shortcut by default, because the downside is quite bad for affected users
  2) think about adding these shortcuts under configuration option, so users like yourself can benefit from the shortcuts 
  3) (bigger job) add system for configuring shortcuts, because everyone has different preferences, and this is not going to be the last time they conflict. It's also very convenient to be able to configure the same shortcuts in OpenMV IDE as I might have in other IDEs.

kwagyeman, 2021-08-30:
QtCreator actually already has #3. I just hid it. I'll renable it and turn the shortcut off by default.

fickludd, 2021-08-30:
Great!

Yeah I think you're keeping a small but good feature-set in general. The only features I really miss from e.g. IntelliJ is `goto definition/usage` and `breadcrumbs` (meaning goto previous/next cursor location).

kwagyeman, 2021-08-30:
The IDE has goto previous location. But, I hid it too. I can re-enable.

kwagyeman, 2022-02-18:
Just disabled these QStringLiteral("Ctrl+Alt+%1"); shortcuts.

closed 2022-02-18 by kwagyeman