openmv/openmv-ide #135: Crash when flashing OpenMV firmware on arduino nano ble 33

Joymaker, 2021-11-19:
Very much like #134 except that, at the critical moment, OpenMV IDE crashes. 
Using OpenMV IDE to follow these instructions: https://docs.arduino.cc/tutorials/nano-33-ble/getting-started-omv

Crashed with the MacOS crash report attached.  MacOS Big Sur 11.6, OpenMV IDE 2.8. Below is pictured the situation just before I pressed the okay button, which led to an instant crash.
[image]

[New bootloader softdevice log.txt](https://github.com/openmv/openmv-ide/files/7571250/New.bootloader.softdevice.log.txt)

[Second crash log.txt](https://github.com/openmv/openmv-ide/files/7571254/Second.crash.log.txt)

kwagyeman, 2022-02-18:
Hi, looking at fixing this now. But, I can't reproduce... Mine works as expected.

kwagyeman, 2022-02-18:
Found why this crashes the IDE. Fixed.

closed 2022-02-18 by kwagyeman

OKPersson, 2022-07-17:
This issue persists on version 2.9.2 on macOS 12.4 Monterey, Thought it was due to the softdevice or bootloader of the nano33 BLE sense, but after a reflash it's still not working.