openmv/openmv-ide #142: Enlarging preview pane makes bottom bar disappear

kdarras, 2022-03-03:
When enlarging the image preview pane vertically, in such a way that it hides the histograms, the bottom bar (which displays the serial terminal button, firmware version, etc.) disappears outside of the screen frame:
[image]

This behavior was consistently observed on a laptop running Ubuntu 20.04 with a Full HD resolution screen. Interestingly when capturing the current window frame with the built-in screenshot app, the entire window will be captured.

kwagyeman, 2022-03-03:
I have to update the IDE framework to fix this. It's due to scaling issues.

Will fix.

kdarras, 2022-04-22:
Unfortunately, this issue was either solved only on my larger screen display or it was re-broken with the latest update... can't remember.
In any case, the issue remains on 2.9.0 on Ubuntu 20.04.4 LTS at present. THis is my whole screen:
[image]

kwagyeman, 2022-04-24:
Hi, I have to update the IDE qt version to fix this. I will get this done soon.

kwagyeman, 2023-07-08:
Ibrahim confirmed this is fixed with the latest IDE.

closed 2023-07-08 by kwagyeman