openmv/openmv-ide #143: button for connecting goes through ON-OFF-ON upon plugging board

kdarras, 2022-03-03:
When I plug in a board to my computer with a USB cable, the button for connecting the board always immediately appears:
[image]
then disappears:
[image]
and then reappears:
[image]
within four seconds or so.

This probably indicates indeed that the camera will be ready for connection for a very short time at the beginning?

The problem is for users who do not know they should only connect to the camera the second time: if they click too early when the button appears for the first time, no openMV cam will be found and the user will be erroneously prompted to go through re-programming as the IDE suggests the board is bricked. Annoying from a user experience point of view as I've had several users doing exactly that.

Running Ubuntu 20.04.

kwagyeman, 2022-03-03:
Hmmm, this is happening because Linux is adding and then removing the device from the device tree.

I'm not really sure what the fix here for this is. The only thing I can do is just add a longer delay... but, this just affects others negatively. Is there any weird hardware you are using on your PC?

kdarras, 2022-03-04:
no weird hardware, only Ubuntu 20.04 straight out of the box. However, the device appears only once inside the "Files" app, and this corresponds to the second time the button shown above appears.

iabdalkader, 2022-03-10:
The first time it detects a device that's the bootloader, which also uses USB CDC. If you run `dmesg -w` and connect the cam, you'll see 2 devices enumerate in a row, the first one is the bootloader. The fix would be to check the serial number in the USB descriptor, the bootloader uses `000000000011` and ignore it. All the other devices use different serial numbers.

kwagyeman, 2022-04-24:
Done

closed 2022-04-24 by kwagyeman